DanielGallo/GivEnergy-Smart-Home-Display #3: No Data in Web Dashboard

BobB001, 2024-08-03:
Hi folks,

Apologies if I'm being thick. I'm trying to set up web access to GivTCP data. I've enabled Web Dashboard in the config and chosen a Web Dashboard port 3042.But when I connect to it I see no data

This is from a dedicated Home Assistant install on a Raspberry Pi.

Any clues:

Thanks in advance.
Bob

DanielGallo, 2024-08-03:
Hi Bob,

Can you open the web browser's Developer Tools, and go to the Console tab - please share details of any errors shown in there after refreshing the page.

Also, share the contents of the `app.json` file, which you should be able to access by navigating to `http://homeassistant:3042/app.json`, based on the URL in the screenshot.

BobB001, 2024-08-04:
Hi Daniel,

Thank you so much for getting back to me. The console tab shows:
[image]
The app.json file, accessed using ```http://homeassistant:3042/app.json``` shows:
[image]

Other info:
I can't see any obvious errors in the GivTCP Logs in Home Assistant
My inverter is ye olde worlde Gen 1, Firmware: D0.450-A0.451. I only have one inverter.
I set the Inverter 1 - Is it on old firmware? toggle to ON in the GivTCP configuration settings
Also the app.json file does not appear to be being refreshed. These values have been the same for at least 1h.

Hope this helps. Thanks again for your super-speedy reply.
Bob

DanielGallo, 2024-08-04:
Please set the export rate (in the GivTCP config) to `0.15` instead of `.15` - this is causing a syntax error when the web dashboard is trying to parse the settings.

DanielGallo, 2024-08-04:
You'll then need to verify the `app.json` has updated once you make the change to the GivTCP config (i.e. make sure the `exportRate` is `0.15`. If it doesn't update, you'll need to find a way to force GivTCP to recreate the web dashboard - I don't recall how this recreation process is handled via Home Assistant unfortunately.

BobB001, 2024-08-04:
Well look at that!
[image]
You superstar.
Thanks Daniel.

closed 2024-08-04 by BobB001

DanielGallo, 2024-08-04:
Ok great! Out of interest, where did you install GivTCP from, or how did you install it? It looks like you might be running an old version of the web dashboard. There should be a version of GivTCP with a newer version of the web dashboard.

BobB001, 2024-08-04:
Hi Daniel,

_> It looks like you might be running an old version of the web dashboard_

... that's an interesting comment. This is a very new install. I downloaded it directly from Home Assistant, following the guidance in a speak-to-the-geek youtube video. Looking back through my install notes, it appears I linked to a different github repository to yours: `https://github.com/britkat1980/giv_tcp`
Having said that, the Web Dashboard on that page looks different to mine.
Could it be something to do with my Gen1 inverter being old?

I'm new to this GivTCP lark and still learning. I've got another question as well, but I'll open a new issue for that. It's a low-priority question.

Thanks again for your help.