DanielGallo/GivEnergy-Smart-Home-Display #5: punycode depreciation

britkat1980, 2024-10-01:
v3 of GivTCP uses Alpine3.19 and is now giving a depreciation warning for punycode module:

```
(node:63) [DEP0040] DeprecationWarning: The `punycode` module is deprecated. Please use a userland alternative instead.
(Use `node --trace-deprecation ...` to show where the warning was created)
```

DanielGallo, 2024-10-01:
@britkat1980 - I've made this pull request: https://github.com/britkat1980/giv_tcp/pull/259. This is to replace the `serve` npm module with `http-server` (to resolve the `punycode` error).
 
I also fixed the `WebDashboard` submodule - it was pointing to an outdated/nonexistent commit, causing it to render a really old version of the web dashboard.

I did a test build of GivTCP image locally, and the web dashboard renders with `http-server` but I couldn't fully test this - ports 6345, 6365, etc. for the inverter data aren't exposed by default. I also had to add port 3000 as being exposed so I could reach the web dashboard. Perhaps you can fully test it on your end?

britkat1980, 2024-10-04:
Thanks, that's working, but I get a different depreciation error:

(node:63) [DEP0066] DeprecationWarning: OutgoingMessage.prototype._headers is deprecated
(Use `node --trace-deprecation ...` to show where the warning was created)

There's couple of tweaks needed to work with GivTCP v3, but I'll create a PR back with that shortly.

willholley, 2024-10-18:
Is there any need to use node at runtime at all? If nginx can be used to serve the static site instead that removes the need to have node/npm in the runtime image completely and knocks a few hundred MB off the size.

I knocked up a quick prototype locally - happy to PR if there's interest.

britkat1980, 2024-10-18:
Always happy to slim down the image size.  Send a PR over on the GivTCP repo

DanielGallo, 2024-10-18:
The dashboard just needs something simple to render the static web content. @willholley your approach sounds good. Thanks!

willholley, 2024-10-18:
Great - I filed a PR at https://github.com/britkat1980/giv_tcp/pull/283

closed 2025-01-01 by DanielGallo