DanielGallo/GivEnergy-Smart-Home-Display #6: Missing battery details in advanced view

yavinmaster, 2024-10-10:
After running V2 for some time, I finally moved to V3 (3.0.4 beta) last night.
I have a V2 hybrid inverter with two 9.5kWh batteries.
On the advanced view of the web interface under the status, I have the outline of only one battery with "undefined% NaN kWh" text in it where I see from the example images I should have two battery outlines with percentages and kWh values.
Everything else appears to be running fine in GivTCP itself although it's only been running for 12 hours.

yavinmaster, 2024-10-25:
Now on the newly released full version 3.0.4 which appears to have solved many issues but this one is still there for me.

yavinmaster, 2024-11-08:
Do you need any more information from me to review this?

DanielGallo, 2024-11-09:
Hi! The data structure of the batteries appears to have changed in GivTCP v3. I've updated the web dashboard to cater for this new data format and submitted a fix to the GivTCP project: https://github.com/britkat1980/giv_tcp/pull/298.

closed 2024-11-09 by DanielGallo