DanielGallo/GivEnergy-Smart-Home-Display #7: No Data on Web when running GivTCP with Gateway and AIO Inverter

adermon, 2024-10-20:
Hi, Ihave GivTCP V3.0.3 running in a docker container with a Gateway and AIO Inverter. I get no data values  on the GUI. The web dashboard is is throwing an exception when getting the Grid_to_Battery sensor value because it doesnt exist in the Gateway data. Im new to GivTcp config and want the Gateway in the configuration but the dashboard assumes that all discovered devices are inverters and will have the same data structure.

[image]

cdrajb, 2024-10-21:
Similar but not the same, I also have all zero readings.
I have an AC3, docker container running 3.0.3, and I get:
Uncaught (in promise) TypeError: Cannot read properties of undefined (reading 'invertor')
    at Helpers.getPropertyValueFromMapping (helpers.js:12:22)
    at App.onResponse (app.js:154:32)
    at app.js:139:16

AberDino, 2024-10-28:
I'm on GivTCP 3.0.4, and my dashboard has no / zero values as well. With the Chrome web browser in debug mode, I get the following error:
`VM30:7 Uncaught (in promise) SyntaxError: Expected ',' or ']' after array element in JSON at position 68 (line 7 column 3)
Promise.then		
App	@	app.js:63
(anonymous)	@	app.js:647`

Interestingly, as you can see in the attached screenshot, data is flowing through (e.g. solarRate), so perhaps once the JSON error is resolved, the dashboard would show values.
[image]

adermon, 2024-10-28:
@AberDino Givtcp writes the dashboard app.json at startup after it has determined the number/config of inverters. The posted json is invalid so maybe there is a bug in 3.0.4. My issue is on givtcp 3.0.3, this has valid app.json but the dashboard doesn't handle GW and AIO correctly. I think my issue would be resolved if I configured givtcp for only the AIO, but I use the GW sensors in HA.

AberDino, 2024-10-28:
@adermon, I 'fixed' the app.json file by adding a comma, and I now have the same error as you:
`helpers.js:12 Uncaught (in promise) TypeError: Cannot read properties of undefined (reading 'Grid_to_Battery')
    at Helpers.getPropertyValueFromMapping (helpers.js:12:22)
    at app.js:179:42
    at Array.forEach (<anonymous>)
    at app.js:178:35
    at Array.forEach (<anonymous>)
    at App.onResponse (app.js:162:17)
    at app.js:139:16`

DanielGallo, 2024-10-28:
Based on the error you're getting, it looks like the underlying data format may have changed slightly in GivTCP 3.x. I'll take a look soon.

DanielGallo, 2024-11-09:
I've proposed a fix for the `app.json` syntax issue in GivTCP v3: https://github.com/britkat1980/giv_tcp/pull/299.

@AberDino @adermon - I don't have a Gateway, so I'm unable to test the web dashboard with one unfortunately. If you can share a full JSON response from GivTCP so I can see the data structure (i.e. call the appropriate `giv_hostname:6345/runAll` endpoint), I can attempt to add support to the web dashboard.

AberDino, 2024-11-10:
Hi DanielGallo, full JSON output from my gateway attached: [gateway.json](https://github.com/user-attachments/files/17690972/gateway.json).

Thanks for looking into this. I have an older tablet, and I'm thinking I could use it to display the GivEnergy Dashboard in the kitchen for example, to try and raise awareness among family members.

DanielGallo, 2024-11-24:
@AberDino @adermon - Hi! I've updated my web dashboard to add basic support for Gateways. I'd appreciate if you could help test this before I push it to the GivTCP project, please. 

If you download the source code from https://github.com/DanielGallo/GivEnergy-Smart-Home-Display/tree/master to your local computer, update the `app.json` file to match the one issued by your GivTCP instance, then run this as a website locally on your computer. 

One way of running this as a website locally is by installing [Node.js](https://nodejs.org/en/download), then from a command prompt / terminal window, install `serve` by running this command: `npm install --global serve`. 
Restart the command prompt / terminal window and `cd` to the web dashboard folder, and run `serve`. This should run the web dashboard as a website on port 3000 of your local computer.

In your desktop web browser, navigate to `http://localhost:3000/?Hostname=10.0.0.210`, making sure to replace `10.0.0.210` with the IP address where your GivTCP is running. This separate instance of the web dashboard will attempt to read the live data from GivTCP.

You should see a "Gateway" summary panel, with "Mode" and "State" values - it should show "On grid" and "Static" as the respective values. I'm not sure whether this information is useful, or whether these values change when you are isolated from the grid? Are there any other values from the Gateway JSON file that you feel would be useful to show in this summary panel?

Regarding the AIO - there are no individual State of Charge percentage values for the battery cells from GivTCP, so I've configured the web dashboard to hide the individual battery percentage/capacity icons if it's an AIO.

Please let me know how you get on.

adermon, 2024-11-24:
@DanielGallo @AberDino Hi i tried that and got a bit further, the dashboard is displaying some items. Below is the web display and the exception from the console. I appear to have a different error now when hitting the rest api endpoint for the Gateway, not sure when this started. I get a response of
{
"result": "No register data cache exists, try again later"
}
the AIO end point is fine

[image]
[image]

DanielGallo, 2024-11-24:
@adermon Thanks. The "no register data cache exists" is an error from the GivTCP side. Have you updated GivTCP over the past few weeks? 

Regarding the AIO - perhaps the data format has changed again. I tested the web dashboard on a version of the JSON provided by @AberDino and it worked fine. Please could you let me know when you last updated GivTCP, and can you provide the JSON files for the gateway (if possible) and AIO? You could try restarting GivTCP to see if that helps resolve the gateway "no register data cache" error.

AberDino, 2024-11-24:
Hi @DanielGallo and @adermon, I just got round to trying this, and I'm very pleased to report it is working for me (I'm on GivTCP 3.0.4). @DanielGallo, thank you for the detailed instructions, that was really helpful for me as non-programmer, and of course also for taking the time to update the code!

Here's a screenshot of my dashboard:
[image]

In terms of the gateway, the "Mode" is definitely useful as it would let the end user know when there is a power cut, at which point keeping an eye on the state of charge is of course more important. I don't know what 'State' represents. I guess you could potentially remove it for now until somebody can explain what it is and whether it would be useful to display.

AberDino, 2024-11-24:
@DanielGallo, slightly off-topic, but just now (with solar power used to power the house and charge the battery), the house / load power value is slightly higher and the battery charge power is slightly lower than what I see in Home Assistant. I suspect that might be because the AIO inverter losses (AC to DC and DC to AC) are not included, i.e. are you using reported battery charge / discharge power, rather than inverter power? I see that the AIO reports "AC charge power" (which matches the inverse of "Inverter Power" when the battery is charging), but unfortunately "AC discharge power" does not seem to exist (but that effectively is "Inverter Power" when it has a positive value). I use "Inverter Power" for my "Power Flow Card Plus" in Home Assistant.

DanielGallo, 2024-11-24:
@AberDino I'm glad to hear it worked! Okay, I've removed the "State" field for now. I've also renamed "Mode" to "Status", as that feels like a more meaningful label. I've committed those changes. I'm holding off from submitting them to the GivTCP project though until I've fixed a few more issues with 3-phase Hybrid inverters, and ruled out any more bugs with data from AIO units.

Out of interest, does the value "On grid" change to something else if you isolate yourself from the grid (e.g. by turning off the isolator switch in the electrical cabinet)?

Regarding power calculations - in my very first version of the web dashboard, each of the "kW" values in the circles were exactly as reported by GivTCP, including the house/load, however these values never correctly totalled in the dashboard, most likely due to the DC conversion loss. A while ago I changed the logic so I show the actual reported PV, Battery, and Grid "kW" values from GivTCP, and I ignore the house/load value from GivTCP and calculate this in the dashboard based on all of the other values. This ensures they always total up correctly, which is more aesthetically pleasing! It's also more complicated in a multi-inverter environment, because other inverters on the same phase can see other inverters as "load" (as they aren't aware of each other), so this was another reason for ignoring the "load" values provided by GivTCP.

This is how I calculate load power, which has worked reliably in my various tests:

```
let totalSourcePower = inverterData.PV_Power + inverterData.Discharge_Power + inverterData.Import_Power;
let loadPower = totalSourcePower - inverterData.Charge_Power - inverterData.Export_Power;
```

If you're seeing a big deviation in values being shown in the web dashboard, feel free to share some updated JSON samples when you have power going to/from various sources (e.g. PV to house and to battery), and I can take a look.

Bananahouse165, 2024-11-27:
Hi, I have a problem with no data, the MQTT devices are populated with data, but its not showing on the web interface.

[image]

DanielGallo, 2024-11-28:
@Bananahouse165 Hi! The issue you're seeing is most likely fixed in the update I mentioned above, although this fix hasn't made its way into the GivTCP version of the web dashboard yet. 

If you feel comfortable following the steps outlined in the following comment, you can test the new version of the web dashboard locally on your computer, to see if it works with your AIO: https://github.com/DanielGallo/GivEnergy-Smart-Home-Display/issues/7#issuecomment-2495723242.

Please let me know if it works, or if you encounter any errors. I don't have an AIO to test, so can't verify myself, but based on the comment from @AberDino above the fix should help the web dashboard work with AIO units.

AberDino, 2024-11-30:
Hi @DanielGallo, just following up on your response from last weekend.

In terms of the "on grid" status, I've not been able to test yet whether that changes when the grid connection drops out, but I would hope it does. I'll let you know as soon as I've tested it.

With regards to the power calculations, I agree it makes sense to calculate the house load. However, I'm wondering, for the AIO at least with the built-in inverter, whether it is correct to use the reported "Charge_Power" and "Discharge_Power" values, as it appears these values do not include the DC/AC conversion losses and the inverter losses. Within the JSON file, there is also an entity "Invertor_Power" (not that both "invert**e**r" and "invert**o**r" seem to be used in the JSON file!), and for me, when I use that value, the calculated house load is pretty close to the real measured house value (I have a second, independent CT clamp arrangement in my consumer unit).

As I mentioned previously, the "Invertor_Power" value goes negative when power is flowing to the AIO, i.e. when it is charging. (It then matches the reported "AC_Charge_Power" value (but this value is positive).) When power is flowing from the AIO, the "Invertor_Power" value is positive.

I've had a look at your JavaScript files to see if I could easily override the "Charge_Power" and "Discharge_Power" definitions, but as a non-programmer I got lost pretty quickly. With minimal change to the code, i.e. still use "Charge_Power" and "Discharge_Power" in the code itself but override the values from a different source, could I try to use "Invertor_Power" for the power calculations, e.g. when "Invertor_Power" is negative, "Charge_Power" is the inverse of "Invertor_Power" and "Discharge_Power" is zero, and when "Invertor_Power" is positive, "Charge_Power" is zero and "Discharge_Power" equals "Invertor_Power"? Let me know what you think, thanks!

abbottcam, 2024-12-19:
Hi Daniel
Is it possible to run this from my home assistant which is also running GivTCP? do does it have to run from a different computer?

DanielGallo, 2024-12-28:
@abbottcam Hi! An older version of the web dashboard is included with GivTCP, but the fixed version mentioned above needs to be used separately at the moment (e.g. on a different computer, or run as a separate process outside of Home Assistant), as the fixes haven't been added into the GivTCP distribution of the web dashboard.

DanielGallo, 2024-12-28:
@AberDino Apologies for the delayed response - life has been a bit hectic over the past month! 

Looking at my own inverter values, it appears `Discharge_Power` could be the actual discharge rate from the batteries, and `Invertor_Power` is the value that takes into account DC to AC losses. For example, this evening I'm seeing a `Discharge_Power` of `3017` but the `Invertor_Power` shows as `2759`.

If you wanted to "correct" the load power shown on the house, you could add `Invertor_Power` as a new sensor in `sensors.js` (notice I add it with a corrected display name of "Invert**e**r_Power"):

```
{
    id: 'Inverter_Power',
    mapping: 'Power.Power.Invertor_Power',
    combinator: {
        singlePhaseSingleInverter: CombinatorType.Any,
        singlePhaseMultipleInverters: CombinatorType.Addition,
        multiplePhases: CombinatorType.Addition
    }
},
```

Then, in app.js, change the following lines:

```
let totalSourcePower = inverterData.PV_Power + inverterData.Discharge_Power + inverterData.Import_Power;
let loadPower = totalSourcePower - inverterData.Charge_Power - inverterData.Export_Power;
```

To this:

```
let dischargePower = 0;
let chargePower = 0;

if (inverterData.Inverter_Power > 0) {
    dischargePower = inverterData.Inverter_Power;
}

if (inverterData.Inverter_Power < 0) {
    chargePower = inverterData.Inverter_Power * -1;
}

let totalSourcePower = inverterData.PV_Power + dischargePower + inverterData.Import_Power;
let loadPower = totalSourcePower - chargePower - inverterData.Export_Power;
```

The battery kW value will be slightly off (as it's still using `Battery_Power`), but the load power on the house should be corrected.

**Note:** I've not tested this when there is solar being generated though - I believe the above code will most likely incorrectly double-count the power when there is solar (e.g. if there is `PV_Power`, it will probably be included in `Invertor_Power`). In which case you would probably need to subtract `PV_Power` from `Invertor_Power` (or something like that). This may have been one of the reasons I chose to use `Discharge_Power` and `Charge_Power`.

AberDino, 2024-12-29:
Thanks @DanielGallo, that is perfect, the house load now matches what I see in HA. There is a little bit of solar output at the moment, and I can confirm it's not included in `Invertor_Power`, however, the solar PV power is just flowing to the house and not used to charge the battery, so I'll have another look when solar output is also used to charge the AIO. I think your code's logic is still valid, we would just be using AIO AC input and output, rather than what appears to be, DC input and output (so excluding AIO inverter losses).

[image]

abbottcam, 2024-12-31:
> @abbottcam Hi! An older version of the web dashboard is included with GivTCP, but the fixed version mentioned above needs to be used separately at the moment (e.g. on a different computer, or run as [...]

might be a silly question,  'update the app.json file to match the one issued by your GivTCP instance' where do I find the one issued by GivTCP?

DanielGallo, 2024-12-31:
@abbottcam Hi! Not a silly question, don't worry. The web app built in to GivTCP typically runs at port 3000, so if the IP address of your GivTCP instance is `10.0.0.100`, you should be able to access the `app.json` config file by navigating to `http://10.0.0.100:3000/app.json` in your web browser.

abbottcam, 2025-01-01:
> might be a silly question, 'update the app.json file to match the one issued by your GivTCP instance' where do I find the one issued by GivTCP?

Happy New Year!
I got it working, the app.json layout was slightly different from the one running on my separate computer. 

This is from GivTCP
{
  "givTcpHosts": [
  {
    "name": "GivTCP",
    "port": "6345"
  }
  {
    "name": "GivTCP2",
    "port": "6346"
  }
  ],
  "solarRate": 0.267,
  "exportRate": 0.15
}

vs from the computer running the dashboard.

{
  "givTcpHosts": [{
    "name": "GivTCP",
    "port": "6345"
  }, {
    "name": "GivTCP2",
    "port": "6346"
  }],
  "solarRate": 0.267,
  "exportRate": 0.15
}

DanielGallo, 2025-01-01:
@AberDino FYI, I've made some fixes to my web dashboard codebase. I remembered that you only have the one inverter, however because you have a gateway and an inverter, my app was incorrectly treating these as 2 inverters - it now only sees this as one inverter, and in doing so will ignore the custom load calculation logic (this calculation logic is only needed when there are multiple inverters on a single phase). So if you grab the latest web dashboard code, it should just use the house load values provided by GivTCP for your single inverter setup. Also, during my testing, solar charge was being included in Inverter Power, so it's best not to use that number.

DanielGallo, 2025-01-01:
New web dashboard has been submitted to the GivTCP project: 
https://github.com/britkat1980/giv_tcp/pull/325
https://github.com/britkat1980/giv_tcp/pull/326

closed 2025-01-01 by DanielGallo

AberDino, 2025-01-04:
> So if you grab the latest web dashboard code (https://github.com/DanielGallo/GivEnergy-Smart-Home-Display), it should just use the house load values provided by GivTCP for your single inverter setup.

Thanks @DanielGallo, that's working fine.