DanielGallo/GivEnergy-Smart-Home-Display #8: no data in dashboard. 3 phase inverter

LeeFarm11, 2024-11-16:
Hi. I have never had any data in the GivTCP web dashboard. I have run the 6345/runAll and file is attached.

I am now running the latest GivTCP Dev3.0.44

Environment details
Inverter - GIV-3HY-11. 3 phase
Batteries - 19.94kWh
Standard HAOS in VMWare

Results are the same at  :8099/dashboard/ and at :3000

[image]

[6345.json](https://github.com/user-attachments/files/17788071/6345.json)

DanielGallo, 2024-11-24:
Thanks for the sample JSON file. I'll take a look soon.

DanielGallo, 2024-12-28:
@LeeFarm11 Hi! I've finally had a chance to take a closer look at this. 

Out of interest, have you updated GivTCP again recently, or is 3.0.44 still the latest version available to you? The reason I ask is because your JSON file contains some odd-looking values, so I'm wondering if they are fixed by a newer version of GivTCP.

For example, these are the "Load Power" values (in Watts):

```
"Load_Phase1_Power": 30.5,
"Load_Phase2_Power": 6530.2,
"Load_Phase3_Power": 6545.4,
"Load_Power": 618.0,
```

I would expect the total of the Phase 1, 2, and 3 values (13,106 Watts) to match the total "Load_Power", but they don't. So I'm not sure whether to use "Load_Power" in my dashboard, or total up the power of the 3 phases and ignore "Load_Power".

There are also other inconsistencies, e.g. `Grid_Apparent_Power` shows as `1009` (Watts), but this doesn't tie-up with the total load power, or the load powers of the individual phases above (when taking into account the other sources of power such as `Battery_Discharge_Power`, `PV_Power`, etc).

There are also no "Flow" values present in the JSON file, so I can't easily map where power is flowing from and to (e.g. solar to battery, battery to house, battery to grid, etc). E.g. the following values are not present in your JSON file (these are present in single-phase inverters, such as single-phase AIO, AC, and Hybrid inverters):

```
  "Power": {
    "Flows": {
      "Battery_to_Grid": 0,
      "Battery_to_House": 0,
      "Grid_to_Battery": 0,
      "Grid_to_House": 1631,
      "Solar_to_Battery": 0,
      "Solar_to_Grid": 0,
      "Solar_to_House": 0
    }
  }
```

I'm not sure whether this omission of the "Flows" data is a bug/error on the GivTCP side, or if it was purposefully omitted due to it being a 3-phase inverter. 

I'm happy to make the web dashboard compatible with 3-phase inverters, but I'm struggling to understand which "load" values to use from GivTCP and how to calculate the power flows between the different sources (battery, solar, grid), as this data isn't being rendered by GivTCP.

LeeFarm11, 2024-12-28:
Hi @DanielGallo I think we are then both a little in the dark.

I haven't had much chance recently to look at much, but most of what you posted is factual information and errors relating either to the Inverter or GivTCP. I wonder if @britkat1980 is able to respond to some of your observations?

Thanks.

DanielGallo, 2024-12-28:
Thanks @LeeFarm11. Yes, happy to hear if @britkat1980 has any thoughts on the above.

trevordowling, 2025-01-12:
Hi, I've got the same issue, also a three phase system.

Stepping through the debug on the browser, it falls over when trying to obtain the variable

**power.power.grid_power**

in static getPropertyValueFromMapping(obj, mapping) in HELPERS.JS

The Obj array is populated with other values that have been pulled from the inverter.

Maybe this will help - I'm no JS expert, so I could be completely wrong, but the Obj returned is NULL, on which the code dies

trevordowling, 2025-01-14:
Hi @DanielGallo - I'll do my best to pull this out for you. Will look at is this week. Big thanks for your help on these items too...

Trevor

trevordowling, 2025-01-15:
Hi Daniel, 

As I'm new to this (I do have a rusty software background) I just thought I'd check that this was what you were looking for (see attached) 

MQTT log and matching GivEnergy data.

If it is then I can generate more files, however I did look at the dump from all of the MQTT activity and there are a lot of errors (mostly the same) that maybe are worth addressing first. It looks like there are requests for data types that are not within the MQTT entitiy. 

Note: values may not align as time between reading was a few minutes (I'll get them much closer in time if this is useful)
[TD2335G007 Data Download 2025-01-15.csv](https://github.com/user-attachments/files/18429714/TD2335G007.Data.Download.2025-01-15.csv)
[home-assistant_mqtt_2025-01-15T19-51-45.509Z.log](https://github.com/user-attachments/files/18429716/home-assistant_mqtt_2025-01-15T19-51-45.509Z.log)
[mqtt-01JHAPZWM4CVGAF0JZ13ET1KH0-GivTCP Meter Details-4afa5b907d64f895efcad7d8ff143a8a.json](https://github.com/user-attachments/files/18429717/mqtt-01JHAPZWM4CVGAF0JZ13ET1KH0-GivTCP.Meter.Details-4afa5b907d64f895efcad7d8ff143a8a.json)

DanielGallo, 2025-01-15:
Hi Trevor,

I actually just need the JSON data rendered by GivTCP via the `runAll` endpoint, for example if GivTCP is running at `10.0.0.200`, it is most likely rendering its data via `http://10.0.0.200:6345/runAll`.

You can also open up my web dashboard in the browser, bring up the browser console and type in `App.downloadInverterData()`, hit Enter, and the browser *should* offer to save the file. Although the web dashboard isn't able to show the data, it would still have been fetched and held in-memory.

So I just need a series of the above JSON files (captured at different times, based on different loads from various sources), plus screenshots from the GivEnergy app captured around the same time (just so I can see what the general numbers look like there).

Hope that helps.

Thanks!

trevordowling, 2025-01-15:
Ok great, I'll have a go tomorrow at getting this together...

Thanks

Trevor

trevordowling, 2025-01-24:
Hi Daniel, first day with some sun, so I've captured some information for your review, hopefully this is useful. 

I've included the console output, as this has errors included while running the App.downloadInverterData()

Excel File is a data dump from GivEnergy Cloud

[TD2335G007 Data Download 2025-01-24 v3.xlsx](https://github.com/user-attachments/files/18534568/TD2335G007.Data.Download.2025-01-24.v3.xlsx)
[House Background + workshop load + solar charging + grid to battery charging v2.json](https://github.com/user-attachments/files/18534571/House.Background.%2B.workshop.load.%2B.solar.charging.%2B.grid.to.battery.charging.v2.json)
[House Background + workshop load + solar charging + grid to battery charging.json](https://github.com/user-attachments/files/18534570/House.Background.%2B.workshop.load.%2B.solar.charging.%2B.grid.to.battery.charging.json)
[House Background + workshop load + solar charging.json](https://github.com/user-attachments/files/18534569/House.Background.%2B.workshop.load.%2B.solar.charging.json)
[House Background + oven + solar charging.json](https://github.com/user-attachments/files/18534572/House.Background.%2B.oven.%2B.solar.charging.json)
[House Background plus solar charging.json](https://github.com/user-attachments/files/18534567/House.Background.plus.solar.charging.json)

Thanks

Trevor

DanielGallo, 2025-01-24:
Thanks Trevor, I'll take a look later. Did you also happen to capture some screenshots from the GivEnergy app showing the power flows between the different sources?

trevordowling, 2025-01-24:
I just grabbed the data - but happy to also grab some screen shots, but the HOME app on my phone does not work (GivEnergy team aware of issues with their app and Three Phase inverters. 

It does work if I use the AWAY connection, but then the data will be delayed, so not sure how easy it would be to link screen shot with data. 

That said - more than happy to try and see if of use

T

trevordowling, 2025-01-24:
Here are two grabs one has forced charging from the grid applied

[image]
[inverter_0_sample (8).json](https://github.com/user-attachments/files/18535152/inverter_0_sample.8.json)
[image]
[inverter_0_sample (6).json](https://github.com/user-attachments/files/18535153/inverter_0_sample.6.json)

DanielGallo, 2025-01-27:
@trevordowling, @LeeFarm11 - Hi! I've just implemented some changes to my web dashboard that attempt to calculate the various missing "flows" between the different sources/targets based on the house load and varying levels of solar, grid, battery import/export, meaning the dashboard should now be able to show a visualisation of 3-phase inverter data. 

I've tested this against some of the data samples provided above and it appears to look reasonably valid, although there are still some odd looking (overly high) values being returned by GivTCP for `Load_Phase1_Power`, `Load_Phase2_Power`, etc, so I'm not using those values - instead I'm using `Load_Power` (which more closely resembles the house load shown in the screenshots above), along with `Meter_Export_Power` , `Meter_Import_Power`, and a few other values, as those also appear to be more accurate.

If you'd like to try the new version of the web dashboard, follow the steps below:

1. Download the source code from https://github.com/DanielGallo/GivEnergy-Smart-Home-Display/tree/master to your local computer.
2. Update the downloaded `app.json` file to match the one issued by your GivTCP instance - if the current web dashboard is running at `http://10.0.0.210:3000`, the `app.json` will be accessible at `http://10.0.0.210:3000/app.json`.
3. Run the downloaded web dashboard code as a website locally on your computer, or somewhere on your local network. One way of running this as a website locally is by installing [Node.js](https://nodejs.org/en/download), then from a command prompt / terminal window, install `serve` by running this command: `npm install --global serve`. Restart the command prompt / terminal window and `cd` to the web dashboard folder, and run `serve`. This should run the web dashboard as a website on port 3000 of your local computer.
4. In your desktop web browser, navigate to `http://localhost:3000/?Hostname=10.0.0.210`, making sure to replace `10.0.0.210` with the IP address where your GivTCP instance is running. This separate instance of the web dashboard will attempt to read the live data from GivTCP and render it.

Please let me know if you have any feedback or spot any issues with the data being rendered.

LeeFarm11, 2025-01-28:
Hi
Sorry for the dumb questions, stuck at step 1. Please could you elaborate? How exactly, and where to exactly?

trevordowling, 2025-01-28:
Hi Daniel / Lee,

I've got this to run and the figures look pretty OK when compared to the GivEnergy Cloud data

Lee - I'm not an expert on GitHub, but I downloaded the code locally, I think you can just download the ZIP, extract it and contiue to follow the instructions,  but I opened this via the GitHub dashboard and Visual Studio Code that I have installed, so from within the GitHub Studio, from the 'Repository' menu, you can open in Visual Studio. I like this solution, but I'm sure there are many other ways to do this.

[image]

trevordowling, 2025-01-28:
Dainel - big thanks for the efforts on this...

I can see other issues with the GivTCP v3 which may be linked to some of the issues we are seeing.

i.e. I have String_1_Voltage & String_2_Voltage shown in the GivEnergy cloud data, but I can't see it within the GivTCP, MQTT, GivTCPPower -  I can only see 'GivTCP Power PV Voltage String 1' variable, there is no 'GivTCP Power PV Voltage String 2'. 

So though the GivTCP v3 is supporting the 3 phase system, there are still some issues to work through.

When I get soem time, I will attempt to dig into this a little more before reporting/opening a bug report

DanielGallo, 2025-01-28:
@LeeFarm11 The quickest way to get the code is to go here: https://github.com/DanielGallo/GivEnergy-Smart-Home-Display, click on the green "Code" button, and "Download Zip". Unzip to a folder on your local computer. Then follow the remaining steps above. Let me know if you get stuck.

@trevordowling I'm glad to hear it's looking promising! There is indeed an issue in GivTCP with rendering out those PV string 1 and 2 values. After a glance at the code, I can see why - the PV string 2 power value is also mapped to `PV_Power_String_1`, and the same with the Voltage and Current values: https://github.com/britkat1980/giv_tcp/blob/d1d223ab2b7d5b4110c179457f8b8d4755ace48e/GivTCP/read.py#L1605C1-L1610C54. Hence why there are no `..._String_2` values being rendered.

```
        power_output['PV_Voltage_String_1']=GEInv.v_pv1
        power_output['PV_Voltage_String_1']=GEInv.v_pv2
        power_output['PV_Current_String_1']=GEInv.i_pv1
        power_output['PV_Current_String_1']=GEInv.i_pv2
        power_output['PV_Power_String_1']=GEInv.p_pv1
        power_output['PV_Power_String_1']=GEInv.p_pv2
```

I'll try and submit a Pull Request with a fix to the GivTCP project later.

DanielGallo, 2025-01-28:
@trevordowling I've submitted a fix to the GivTCP project for those PV mappings as part of these pull requests: https://github.com/britkat1980/giv_tcp/pull/326 and https://github.com/britkat1980/giv_tcp/pull/325. Also included the latest version of the web dashboard.

closed 2025-01-28 by DanielGallo

trevordowling, 2025-01-28:
> <DanielGallo/GivEnergy-Smart-Home-Display/issue/8/issue_event/16099864986@
> github.com>

Hi Daniel,

Thanks for this, I'll spend some time taking a look at this just to educate
myself a little more around this system. At a quick glance I can see a lot
of codes been written to get this far.

Having coded for a living for a years doing Broadcast video mixers,
Bluetooth and Wi-Fi, I've a genuine appreciation of the effort.

Regards

Trevor

DanielGallo, 2025-01-28:
Thanks Trevor!