DanielGallo/GivEnergy-Smart-Home-Display #9: Individual battery data not appearing correctly on the dashboard

daveyge, 2025-01-18:
Hi Daniel

I have two batteries like the examples you show on your Readme page but although the SOC values appear to be present in the JSON file, there seems to be an error in parsing them when I specify the AdvancedInfo option. 
Screenshot below and I followed the App.downloadInverterData() tip in your other open issue to get the JSON  which is attached.  I am using GivTCP v3.0.4 in HA and Chrome browser v131.0.6778.265 on Windows 11 (please let me know if you need any other info) 

Any advice to resolve this?

Many thanks

David

[image]

[Alldata.txt](https://github.com/user-attachments/files/18464038/Alldata.txt)

DanielGallo, 2025-01-18:
Hi David,

This issue should be resolved in the latest version of the web dashboard. I submitted the fixes to the GivTCP project a few weeks ago, but they've not been included yet.

If you'd like to try the new version of the web dashboard in the meantime, follow the instructions below: 

1. Download the source code from https://github.com/DanielGallo/GivEnergy-Smart-Home-Display/tree/master to your local computer.
2. Update the `app.json` file to match the one issued by your GivTCP instance - if the current web dashboard is running at `http://10.0.0.210:3000`, the `app.json` will be accessible at `http://10.0.0.210:3000/app.json`.
3. Run the downloaded web dashboard code as a website locally on your computer, or somewhere on your local network. One way of running this as a website locally is by installing [Node.js](https://nodejs.org/en/download), then from a command prompt / terminal window, install `serve` by running this command: `npm install --global serve`. Restart the command prompt / terminal window and `cd` to the web dashboard folder, and run `serve`. This should run the web dashboard as a website on port 3000 of your local computer.
4. In your desktop web browser, navigate to `http://localhost:3000/?Hostname=10.0.0.210`, making sure to replace `10.0.0.210` with the IP address where your GivTCP instance is running. This separate instance of the web dashboard will attempt to read the live data from GivTCP, and the batteries should be summarised correctly.

Let me know how you get on.

Thanks,

Daniel

daveyge, 2025-01-19:
Hi Daniel.   Yes sorry,  I only realised that after posting...I should have checked the closed issues first.  Thanks very much for the new version info, really good of you.  I'm out today but will give that a go tomorrow.

Cheers

David

DanielGallo, 2025-01-21:
No problem David! Feel free to re-open this issue if you still spot a problem using the new version.

closed 2025-01-21 by DanielGallo