DanielGallo/GivEnergy-Smart-Home-Display #10: Web Dashboard No Values

AberDino, 2025-04-13:
Hi @DanielGallo, sorry to bother you again.

Referring to [this issue](https://github.com/DanielGallo/GivEnergy-Smart-Home-Display/issues/7), you pushed some fixes to the GivTCP code.

A few days ago, GivTCP v3.1.3 was released, but it had an issue with single AIO setups where people had both the gateway and AIO configured in GivTCP. Disabling the gateway 'fixed' that, but some people, including myself, like to use some of the data provided by the gateway. GivTCP-BETA v3.1.4 was released this morning and it is working fine for me, with both the gateway and AIO configured within GivTCP.

Unfortunately, the web dashboard in GivTCP-BETA v3.1.4 is not working for me:
```
Uncaught (in promise) TypeError: Cannot read properties of undefined (reading 'Grid_to_Battery')
    at Helpers.getPropertyValueFromMapping (helpers.js:12:22)
    at app.js:179:42
    at Array.forEach (<anonymous>)
    at app.js:178:35
    at Array.forEach (<anonymous>)
    at App.onResponse (app.js:162:17)
    at app.js:139:16
```

I was wondering if your code changes made it into GivTCP-BETA v3.1.4? If so, would you mind having another look at this?

DanielGallo, 2025-04-13:
Hi, no problem! I submitted my changes to the GivTCP project several months ago, but they've not been included yet - the latest dev and beta versions contain a much older version.

I'll try and submit them again.

AberDino, 2025-04-13:
> Hi, no problem! I submitted my changes to the GivTCP project several months ago, but they've not been included yet - the latest dev and beta versions of GivTCP contain a much older version of the web dashboard.
> 
> I'll try and submit them again.

Thanks @DanielGallo, at least that explains the symptoms. Hopefully the changes can be included soon.

DanielGallo, 2025-04-13:
https://github.com/britkat1980/giv_tcp/pull/371
https://github.com/britkat1980/giv_tcp/pull/372

closed 2025-04-13 by DanielGallo