DanielGallo/GivEnergy-Smart-Home-Display #11: GivTCP 3.1.6 Web Dashboard

AberDino, 2025-04-13:
Hi @DanielGallo,

The good news is that it appears that the updated web dashboard has now been merged into the main GivTCP code, as stated in the 3.1.6 changelog. However, for some reason, the dashboard still isn't working for me (same "TypeError: Cannot read properties of undefined (reading 'Grid_to_Battery')" as before). Note that nothing is shown on the right-hand of the page either (gateway status etc.).

When I download the giv_tcp repository and 'serve' the included WebDashboard folder, the data appears fine, including the gateway status. How do I check that the correct web dashboard files have made it onto my HA host, i.e. where are these files stored?

Thanks!

DanielGallo, 2025-04-13:
@AberDino Hi! 3.1.6 (https://hub.docker.com/r/britkat/giv_tcp-ma/tags) doesn't have the latest Web Dashboard files. The message you saw in the changelog within Home Assistant might be referring to my earlier updates from a few months ago that didn't make their way in there successfully.

Perhaps the next version of GivTCP will include the latest Web Dashboard files that were merged into the GivTCP repository earlier today - @britkat1980 should know for sure.

One way to tell when it's running the latest version is by looking at the source code of the Web Dashboard in the browser - `js/sensors.js` will contain a `GatewaySensors` object on line 5.

AberDino, 2025-04-14:
Thanks @DanielGallo for the clarifications, that's really useful, and it means I don't have to keep asking you ;-). Here's hoping the new Web Dashboard files will be included with GivTCP soon...

closed 2025-04-14 by AberDino

britkat1980, 2025-04-14:
I think I did include it in latest release... Can you confirm if its there or not in the current Production release?

AberDino, 2025-04-14:
Hi @britkat1980, not sure if you are asking me, but I have GivTCP 3.1.6 installed, and following @DanielGallo's instructions ("One way to tell when it's running the latest version is by looking at the source code of the Web Dashboard in the browser - js/sensors.js will contain a GatewaySensors object on line 5"), the GatewaySensors object isn't there.

britkat1980, 2025-04-14:
Odd, can you try Dev3.1.8 which i've just pushed, to see if that has the right code in it?

DanielGallo, 2025-04-14:
@britkat1980 Hi! I checked the 3.1.6 master/production release last night and it contained an older version of the WebDashboard code. The GivTCP repo now definitely contains the latest version of the Web Dashboard code though, so maybe there is an issue in the build/deployment process that has a stale copy of the code somewhere.

I can check out the 3.1.8 dev release this evening, thanks!

DanielGallo, 2025-04-14:
Excellent, thanks for checking and confirming!

chrisckc, 2025-04-16:
To get this working I had to add 2 additional port mappings (6345 and 6350) in docker compose because I am not using host networking, looking at the developer console on the web page showed the issue.

```
    ports: # "[host:]container[/protocol]"
      #- "1883:1883"                   # MQTT, exposed by a different container
      - "8098:8098"                   # REST and Config web server (https on 8098)
      - "8099:8099"                   # REST and Config web server
      - "3001:3001"                   # WEB Dashboard, change this to 3001 in config/allsettings.json to avoid conflict with Grafana
      - "6345:6345"                   # WEB Dashboard API
      - "6350:6350"                   # WEB Dashboard API
```