DanielGallo/GivEnergy-Smart-Home-Display #12: Show Advances Info?

mprowe, 2025-04-19:
Thank you for the Dashboard. Quick question, if I may?
What are the two entities I've highlighted mapped to?

[image]

Do I need to tune my config somehow?

Regards, Martin

DanielGallo, 2025-04-22:
This looks like a bug that I fixed in a recent version of the web dashboard. Are you using the web dashboard that's bundled with GivTCP? If so, the fix might not be released on the GivTCP side yet. The fix should be in the dev 3.1.8 version of GivTCP, according to this comment: https://github.com/DanielGallo/GivEnergy-Smart-Home-Display/issues/11#issuecomment-2802555769

DanielGallo, 2025-04-22:
If you are using an AIO unit, the individual battery graphics (where it says "undefined") should be hidden in the fixed version (as the AIO is essentially treated as just one battery).

I also improved the mappings in a recent version of the web dashboard, as the structure of the batteries data changed slightly on the GivTCP side, so if you do have an AC or Hybrid inverter with one or more battery packs, the battery graphics should show the individual state of charge percentages and kWh remaining.

mprowe, 2025-04-23:
Thanks for the advice Daniel,

I am on ver 3.1.6 of GivTCP. So, I'll look forward to the upgrade to ver 3.1.8 when it arrives.
This is my startup log should it be relevant:
```
2025-04-23 06:41:49,915 - startup     -  [INFO    ] - ==================== STARTING GivTCP==========================
2025-04-23 06:41:49,948 - startup     -  [INFO    ] - Network Found: 192.168.x.y/24
2025-04-23 06:41:51,951 - startup     -  [INFO    ] - Searching for Inverters
2025-04-23 06:41:51,951 - startup     -  [INFO    ] - Scanning network for GivEnergy Devices...
2025-04-23 06:41:57,077 - client      -  [INFO    ] - Connection established to 192.168.x.y:8899
2025-04-23 06:41:57,078 - client      -  [INFO    ] - Detecting plant
2025-04-23 06:41:58,321 - client      -  [INFO    ] - Plant Detected
2025-04-23 06:42:01,082 - startup     -  [INFO    ] - Inverter SA2219G385 which is a Gen1 - Hybrid with 1 batteries has been found at: 192.168.x.y
2025-04-23 06:42:01,095 - startup     -  [INFO    ] - Setting up invertor: 1
2025-04-23 06:42:01,172 - startup     -  [INFO    ] - ==============================================================
2025-04-23 06:42:01,172 - startup     -  [INFO    ] - ====             Web Gui Config is at                     ====
2025-04-23 06:42:01,173 - startup     -  [INFO    ] - ====     http://192.168.x.y:8099/config.html             ====
2025-04-23 06:42:01,173 - startup     -  [INFO    ] - ==============================================================
2025-04-23 06:42:01,174 - startup     -  [INFO    ] - Running Invertor 1 (SA2219G385) read loop every 30/120s
2025-04-23 06:42:01,181 - startup     -  [INFO    ] - Serving Web Dashboard from port 3000
[2025-04-23 06:42:01 +0100] [70] [INFO] Starting gunicorn 23.0.0
[2025-04-23 06:42:01 +0100] [70] [INFO] Listening at: http://0.0.0.0:6350 (70)
[2025-04-23 06:42:01 +0100] [70] [INFO] Using worker: sync
[2025-04-23 06:42:01 +0100] [86] [INFO] Booting worker with pid: 86
[2025-04-23 06:42:01 +0100] [80] [INFO] Starting gunicorn 23.0.0
[2025-04-23 06:42:01 +0100] [80] [INFO] Listening at: http://0.0.0.0:6345 (80)
[2025-04-23 06:42:01 +0100] [80] [INFO] Using worker: sync
[2025-04-23 06:42:01 +0100] [87] [INFO] Booting worker with pid: 87
[2025-04-23 06:42:01 +0100] [88] [INFO] Booting worker with pid: 88
[2025-04-23 06:42:01 +0100] [89] [INFO] Booting worker with pid: 89
2025-04-23 06:42:02,802 - GivTCP - read        -  [INFO    ] - Starting watch_plant loop...
2025-04-23 06:42:02,804 - GivTCP - GivLUT      -  [CRITICAL] - Opening Modbus Connection to 192.168.x.y
2025-04-23 06:42:02,807 - GivTCP - read        -  [CRITICAL] - Detecting inverter characteristics...
2025-04-23 06:42:10,382 - GivTCP - read        -  [INFO    ] - Publishing Home Assistant Discovery messages
```

PS.... What is the difference between the Dashboard at port 8099 and port 3000?

closed 2025-04-23 by mprowe