DanielGallo/GivEnergy-Smart-Home-Display #14: Reading the battery state (charging, discharging & idle)

mprowe, 2025-06-30:
Hi Daniel,
I note that in addition to the SOC%, you are able show time-based condition

[image]

Is this sensor available for me to display in my Dashboard?

Regards, Martin

mprowe, 2025-07-02:
Thanks Daniel,
I have been attempting (on & off) to implement your example since April. But not getting very far!
I understand your logic. My problem is integrating the code into HA. I have tried to include it in my configuration.yaml without success? My main effort however, is to test that it works by using the Template Editor in preparation for creating a Helper.
This is what I am seeing:

[image]
I know the problem is in the curly braces and the indentation. But how to correct?

Regards, M.

DanielGallo, 2025-07-21:
Hi Martin,

Sorry for the delayed reply. In your `configuration.yaml`, try defining it like this, ensuring it's placed under the `sensor:` key (you may need to add `sensor:` if it doesn’t already exist in the YAML file):

```
sensor:
  - platform: template
    sensors:
      battery_state:
        value_template: >-
          {% if states('sensor.abc123_discharge_power') | float(0) > 0 %} Discharging
          {% elif states('sensor.abc123_charge_power') | float(0) > 0 %} Charging
          {% else %} Idle
          {% endif %}
        friendly_name: Battery State
```

I have it defined exactly like the above in my `configuration.yaml`, so hopefully this will also work for you.

closed 2025-07-22 by mprowe